SyrianSpock/realsense_gazebo_plugin #1: depth not available

bertcay, 2016-11-14:
Hi,

I was looking for a Gazebo plugin of realsense and tried this project.

When launching the the "demo" here is an output :
[Wrn] [msgs.cc:1655] Conversion of sensor type[depth] not suppported.
--> depth topic is not published.
Any ideas ?

Thanks
Bertrand

SyrianSpock, 2016-11-14:
You are right. Only the RGB image was published over ROS.
faf87332e790cdf078fda2c7a1d4e8f67a05a989 should fix your problem. It should publish the depth image (but no point cloud).
I use Gazebo 7 + ROS indigo, so let me know if that works for you.

bertcay, 2016-11-14:
Thanks for this quick update :)
By "no point cloud" do you mean that output format differs from the real realsense output ?

SyrianSpock, 2016-11-15:
If I remember correctly, there are multiple launch files for the realsense, some of them only publish depth image (2D image where the pixel value is depth), while other would publish point clouds.
I haven't touched the project in two months, I will need to check the current state of the sensor driver to verify that the output format is still the same.

closed 2017-04-11 by SyrianSpock