SyrianSpock/realsense_gazebo_plugin #2: launch error

serhatkilicc, 2017-01-20:
Error [parser_urdf.cc:2608] Unable to call parseURDF on robot model
Error [parser.cc:273] parse as old deprecated model file failed.
Error [parser.cc:614] Unable to read file[/home/try/Desktop/SimulationFiles/navigation/mybot_ws/src/gazeboRS/models/realsense_camera/model.sdf]
Error [parser.cc:688] Error reading element <world>
Error [parser.cc:348] Unable to read element <sdf>
Error:   Could not find the 'robot' element in the xml file
         at line 81 in /build/buildd/urdfdom-0.2.10+dfsg/urdf_parser/src/model.cpp
Error [parser_urdf.cc:2608] Unable to call parseURDF on robot model
Error [parser.cc:273] parse as old deprecated model file failed.
Error [Server.cc:300] Unable to read sdf file[/home/try/Desktop/SimulationFiles/navigation/mybot_ws/src/gazeboRS/worlds/realsense.world]
gzserver: /usr/include/boost/thread/pthread/recursive_mutex.hpp:101: boost::recursive_mutex::~recursive_mutex(): Assertion `!pthread_mutex_destroy(&m)' failed.
Aborted (core dumped)
[gazebo-2] process has died [pid 14248, exit code 134, cmd /opt/ros/indigo/lib/gazebo_ros/gzserver -r --verbose -e ode /home/try/Desktop/SimulationFiles/navigation/mybot_ws/src/gazeboRS/worlds/realsense.world __name:=gazebo __log:=/home/try/.ros/log/d7c27010-df27-11e6-bd79-68a3c43c0c01/gazebo-2.log].
log file: /home/try/.ros/log/d7c27010-df27-11e6-bd79-68a3c43c0c01/gazebo-2*.log

-----------------------------
Hi,

roslaunch realsense_gazebo_plugin realsense.launch  

I get this error when I run the command

SyrianSpock, 2017-01-23:
Hi,
What version of ROS/Gazebo are you using ?

serhatkilicc, 2017-01-23:
Ros indigo
Gazebo 2-2-3

SyrianSpock, 2017-01-24:
You need Gazebo 7 or higher.
Here is a [guide on how to upgrade your Gazebo version](http://gazebosim.org/tutorials?tut=ros_wrapper_versions#UsingaspecificGazeboversionwithROS).

SyrianSpock, 2017-02-03:
I'll close this. Feel free to re-open if it still doesn't work for you.

closed 2017-02-03 by SyrianSpock

priyablue, 2018-04-11:
Hi,
I face the same trouble. But I am using Gazebo 7 version. Can you help me to clear this issue.Thanks in advance.

priyablue, 2018-04-11:
This the error I get ,

Error:   No link elements found in urdf file
         at line 180 in /build/urdfdom-UJ3kd6/urdfdom-0.4.1/urdf_parser/src/model.cpp
Error [parser_urdf.cc:3408] Unable to call parseURDF on robot model
Error [parser.cc:340] parse as old deprecated model file failed.
[INFO] [1523465835.001393, 10.156000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name widowx
[ INFO] [1523465835.011236722, 10.165000000]: Physics dynamic reconfigure ready.
[urdf_spawner-4] process has finished cleanly

SyrianSpock, 2018-05-27:
@priyablue There is now an example to add the sensor to your robot URDF, see https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/urdf/rs200_simulation.xacro

q8wwe, 2020-03-02:
I'm facing same issue , how can we fix it

Fellfalla, 2021-02-17:
When importing via gazebo GUI, I'm getting this error

> Error [parser_urdf.cc:3193] Unable to call parseURDF on robot model
> Error [parser.cc:488] parse as old deprecated model file failed.
> [Err] [ModelMaker.cc:214] No model or light in SDF

Running `gz sdf -p my_model.xacro` outputs just an useless log. Calling `cat ~/.gazebo/gz-11345/gzsdf.log` returns

> Gazebo multi-robot simulator, version 11.3.0
> Copyright (C) 2012 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org