SyrianSpock/realsense_gazebo_plugin #3: Loading from a *.xacro file and parameters list?

doisyg, 2017-02-02:
Hi,
Sorry if this is a trivial question, this is the first time i am using a gazebo ros sensor plugin.
Is there a parameter list of the plugin and how to access it? (i.e. similar to the realsense ros driver paramters which set topics path and activate/desactivate the different color/ir/depth streams).
Additionally, do you have a example of how to load the plugin from a *.xacro file?
Thanks!

SyrianSpock, 2017-02-03:
Hi,

> Is there a parameter list of the plugin and how to access it? (i.e. similar to the realsense ros driver paramters which set topics path and activate/desactivate the different color/ir/depth streams).
Currently there is no dynamic reconfigure interface for the plugin. I don't know if it would be useful, as we are limited in simulation and can't replicate the same physical behavior of the sensor.

> Additionally, do you have a example of how to load the plugin from a *.xacro file?
I have been planning to do this, but didn't have much time to work on this project. For now it's limited to the provided SDF file.

SyrianSpock, 2018-05-27:
Solved by #10

closed 2018-05-27 by SyrianSpock