SyrianSpock/realsense_gazebo_plugin #4: camera_info message for your package

ruijiayang, 2017-03-23:
Hi, I am trying to convert the depth image output by the realsense_gazebo_plugin into a point cloud, using the depth_image_proc package (in ROS's image_pipeline). However, this requires camera_info information. 

Would you be able to add a camera_info topic for the simulated realsense_camera?

If not, can you provide values for the following fields of the sensor_msgs/CameraInfo.msg for this simulated realsense_camera, so that others can add that topic themselves to their clones of your package if they wish?

std_msgs/Header header
uint32 height
uint32 width
string distortion_model
float64[] D
float64[9] K
float64[9] R
float64[12] P
uint32 binning_x
uint32 binning_y
sensor_msgs/RegionOfInterest roi

(Source: sensor_msgs/CameraInfo.msg definition - http://docs.ros.org/api/sensor_msgs/html/msg/CameraInfo.html)

Thank you! This package has been very useful to me so far, and with camera_info added, will be even more so :-)

rachelruijiayang, 2017-03-23:
From the image topics, the Header, height, and width information are already provided, I think. 
- The header can be taken from an image header
- Height is 480, width is 640

```
ruijia@bliss:~/aero_ws/src$ rostopic echo /realsense/camera/color/image_raw
header: 
  seq: 0
  stamp: 
    secs: 21
    nsecs: 230000000
  frame_id: color
height: 480
width: 640
encoding: rgb8
is_bigendian: 0
step: 1920
data: [102, 102, 102, 101, 101, 101, 101, 101, 101, 99, 99, 99, 103, 103, 103, 101, 101, 101, 101, 101, 101, 101, 101, 101, 102, 102, 102, 101, 101, 101, 105, 105, 105, 103, 103, 103, 99, 99, 99, 103, 103, 103, 104, 104, 104, 102, 102, 102, 101, 101, 101, 100, 100, ...
```

Should there be any values for the distortion model, binning, and ROI fields? I don't think so, right?

In this case, it should be very easy to add CameraInfo...

SyrianSpock, 2018-02-20:
Hi, this has been open for a while, I just pushed a fix to address it. There is also a demo for using depth_image_proc package. Can you check if it solves the issue?

closed 2018-05-27 by SyrianSpock