SyrianSpock/realsense_gazebo_plugin #6: Compressed Depth Stream throws Ogre Error in Rviz [Ros Kinetic, Gazebo 7.1]

geoeo, 2018-02-19:
Hello,

I just noticed that when accessing the compressed depth stream in rViz I get the following exception:  OGRE EXCEPTION(2:InvalidParametersException): Stream size does not match calculated image size

Do you know why this happens?

Best,
Marc

SyrianSpock, 2018-02-20:
Hi,

Unfortunately I have never used the compressed streams.
Do you have the necessary ROS packages installed? I think you need `ros-kinetic-compressed-depth-image-transport`.

geoeo, 2018-02-21:
Ah. Thats some place I can start looking. Thanks :)

Until recently it didnt throw an exception but just returned a black image. But for some reason my OGRE references got lost. I really dont know how, and I had to reinstall OGRE.

Since then I noticed this. So It might be related to a OGRE newer version. 

Thanks again!

closed 2018-04-06 by SyrianSpock