SyrianSpock/realsense_gazebo_plugin #9: can't rostopic echo pointcloud with gazebo 9

mfe7, 2018-04-03:
```
$ rostopic echo /realsense/camera/depth_registered/points
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

The nodes are starting up without errors, but I can't echo the realsense topics

SyrianSpock, 2018-04-04:
Can you try running `rosparam set use_sim_time true` and then relaunch the node?

mfe7, 2018-04-06:
`use_sim_time` is indeed set to `true`

No errors when I build:
```
mfe@mfe-ubuntu:~/realsense_ws$ catkin build realsense_gazebo_plugin 
---------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/mfe/ford_ws/devel:/opt/ros/kinetic
Workspace:                   /home/mfe/realsense_ws
---------------------------------------------------------------------
Source Space:       [exists] /home/mfe/realsense_ws/src
Log Space:         [missing] /home/mfe/realsense_ws/logs
Build Space:        [exists] /home/mfe/realsense_ws/build
Devel Space:        [exists] /home/mfe/realsense_ws/devel
Install Space:      [unused] /home/mfe/realsense_ws/install
DESTDIR:            [unused] None
---------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                                                        
[build] Updating package table.                                                                                   
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                               
Finished  <<< catkin_tools_prebuild                  [ 1.4 seconds ]                                              
Starting  >>> realsense_gazebo_plugin                                                                             
__________________________________________________________________________________________________________________
Warnings   << realsense_gazebo_plugin:cmake /home/mfe/realsense_ws/logs/realsense_gazebo_plugin/build.cmake.000.log
CMake Warning (dev) at /usr/local/lib/python2.7/dist-packages/cmake/data/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/local/lib/python2.7/dist-packages/cmake/data/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:309 (find_package)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  CMakeLists.txt:15 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/mfe/realsense_ws/build/realsense_gazebo_plugin; catkin build --get-env realsense_gazebo_plugin | catkin env -si  /usr/local/bin/cmake /home/mfe/realsense_ws/src/realsense_gazebo_plugin --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mfe/realsense_ws/devel/.private/realsense_gazebo_plugin -DCMAKE_INSTALL_PREFIX=/home/mfe/realsense_ws/install; cd -
..................................................................................................................
Finished  <<< realsense_gazebo_plugin                [ 19.5 seconds ]                                             
[build] Summary: All 2 packages succeeded!                                                                        
[build]   Ignored:   None.                                                                                        
[build]   Warnings:  1 packages succeeded with warnings.                                                          
[build]   Abandoned: None.                                                                                        
[build]   Failed:    None.                                                                                        
[build] Runtime: 20.9 seconds total.  
```

When I run the tests, not much seems to happen so I do ctrl+c after 20sec:

```
mfe@mfe-ubuntu:~/realsense_ws/src/realsense_gazebo_plugin$ rostest realsense_gazebo_plugin realsense_streams.test 
[... 5 lines cut ...]
RealSensePlugin: The rs_camera plugin is attach to model realsense_camera
[ INFO]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO]: Physics dynamic reconfigure ready.
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/mfe/.ros/log/rostest-mfe-ubuntu-2623.log
```

SyrianSpock, 2018-05-27:
I could not reproduce this on Ubuntu 18.04 + ROS Melodic + Gazebo 9.
I ran
```
$ roslaunch realsense_gazebo_plugin realsense_urdf.launch
```
then in a second terminal
```
$ roslaunch realsense_gazebo_plugin depth_proc.launch
```
and I get
```
$ rostopic hz /realsense/camera/depth_registered/points
average rate: 20.892
	min: 0.000s max: 0.159s std dev: 0.02522s window: 7242
average rate: 20.894
	min: 0.000s max: 0.159s std dev: 0.02523s window: 7256
average rate: 20.891
	min: 0.000s max: 0.159s std dev: 0.02523s window: 7270
```

Are you still facing this issue?

closed 2018-05-27 by SyrianSpock