SyrianSpock/realsense_gazebo_plugin #14: RViz & GAZEBO not opening up

bluetulip89, 2018-06-06:
I build the catkin command. Running depth image command  in one terminal and realsense command as mentioned in your Readme. But RViz and GAZEBO doesnt open as you have shown the picture. Help me out to clear this error.

bluetulip89, 2018-06-06:
I am using Ubuntu 16.04 ROS Kinetic

SyrianSpock, 2018-06-11:
What is the error message you get?
Also note that you should use [kinetic-devel](https://github.com/SyrianSpock/realsense_gazebo_plugin/tree/kinetic-devel) branch for the package to work on ROS kinetic / Gazebo 7.

bluetulip89, 2018-06-12:
Has used the same. Ok will post the error what i get in the terminal. Kindly help out.

mahty24, 2018-06-12:
I am facing the same issue, and it doen't have errors it only has a warnings as follow.
[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported.

SyrianSpock, 2018-06-12:
You should launch rviz from another terminal, simply run: 
```
rviz
```

And Gazebo is intentionally started headless (no GUI), so if you want to its UI you need to set 
```xml
<arg name="headless" value="false"/>
<arg name="gui" value="true"/>
```

mahty24, 2018-06-12:
thank you @SyrianSpock.

SyrianSpock, 2018-06-18:
Feel free to reopen if you still have this issue

closed 2018-06-18 by SyrianSpock