SyrianSpock/realsense_gazebo_plugin #15: ROSTEST Command fails

bluetulip89, 2018-06-06:
I have pasted the error .Please go through and let me know what is the mistake.

logging to /home/sp/.ros/log/rostest-hp-omen-8339.log
[ROSUNIT] Outputting test results to /home/sp/.ros/test_results/realsense_gazebo_plugin/rostest-test_realsense_streams.xml
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/sp/.ros/log/rostest-hp-omen-8339.log

SyrianSpock, 2018-06-11:
Can you try again with the correct branch: [kinetic-devel](rostest realsense_gazebo_plugin realsense_streams.test)?

SyrianSpock, 2018-06-18:
Feel free to reopen if the issue is still present with the correct branch

closed 2018-06-18 by SyrianSpock