SyrianSpock/realsense_gazebo_plugin #16: Gazebo doesnt openup

priyablue, 2018-07-08:
Hi Syrian, I tried using the same package and cloned it for kinetic branch and when i try running the two commands in different terminal Gazebo doesnt openup. it shows Depth warning as there is some topics not publishing on the same.

Can you help me in clearing this error.

SyrianSpock, 2018-07-08:
The readme is a bit misleading. The Gazebo UI is disabled in [realsense.launch](https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/launch/realsense.launch#L4-L5). If you want to get the Gazebo UI, replace these two lines by
```
    <arg name="headless" value="false"/>
    <arg name="gui" value="true"/>
```

priyablue, 2018-07-09:
Thanks for the response. 

why am i not able to get the pose values been set in URDF. I want to lift the camera bit higher, so i have modified the Z values in realsense-RS200.macro.xacro. But there is no change in the pose of camera. 

Help me out. 

Thanks

Danfoa, 2018-07-09:
@priyablue I believe you have a confusion on URDF/Macro functions or in which file in this repository is what.

If you want to modify the position of the camera with respect to the world or your reference link you have to modify the `xyz` parameter on urdf file that uses the `REALSENSE-RS200. macro. xacro` macro function (in this repository example this file is called rs200-simulation.xacro). Any modification to the macro function referenced before will create changes in the internal structure of the camera, which is something that in the most cases should be avoided.

SyrianSpock, 2018-07-26:
@priyablue Feel free to reopen if you still have problems

closed 2018-07-26 by SyrianSpock