SyrianSpock/realsense_gazebo_plugin #20: Gazebo 7, no function SimTime

amerkay, 2018-09-18:
Quick note, to fix the following `catkin_make` error for Gazebo 7 on Ubuntu 16.04 with ROS Kinetic:
```
/home/amerk/catkin_ws/src/external_lib/realsense_gazebo_plugin/src/RealSensePlugin.cpp:123:42: error: ‘class gazebo::physics::World’ has no member named ‘SimTime’
   this->transportNode->Init(this->world->SimTime()); # and Name()
```

The function names `SimTime()` and `Name()` are now `GetSimTime()` and `GetName()` respectively. 

Thanks for the great library!

SyrianSpock, 2018-09-20:
Indeed, this has been reported previously, so I created the `kinetic-devel` branch to remain compatible with ROS Kinetic + Gazebo 7.
I should have make it explicit in the readme.

SyrianSpock, 2018-09-27:
Readme updated to avoid facing this issue for future users.

Thanks again @amerkay for reporting.

closed 2018-09-27 by SyrianSpock

amerkay, 2018-09-29:
Thank you for that. I could've looked closer. Works like a charm, great job here!

A quick note, I noticed the URDF file has `clip = 100` value for the depth camera:
https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/6fa6c12f812ff2f59451f7a8a8b675e4e0953306/urdf/realsense-RS200.macro.xacro#L185

In reality the range is closer to 8-10 meters. Just if that helps anyone else. It helped with using RTABmap with the Gazebo simulation.