SyrianSpock/realsense_gazebo_plugin #21: The D matrix is empty

abylikhsanov, 2018-09-24:
Getting error:
`Cannot call rectifyPoint when distortion is unknown.`
This is the camera_info topic echo, which shows that the D matrix is empty:
distortion_model: ''
```
D: []
K: [463.8890075683594, 0.0, 320.0, 0.0, 463.8890075683594, 240.0, 0.0, 0.0, 1.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [463.8890075683594, 0.0, 320.0, 0.0, 0.0, 463.8890075683594, 240.0, 0.0, 0.0, 0.0, 1.0, 0.0]
```

SyrianSpock, 2018-09-27:
Indeed, there is no camera distortion applied to the sensor simulated here. So there is no need to rectify the images published here. 

If you would like to add distortion to the simulated model, you modify the [sensor definition file](/https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/urdf/realsense-RS200.macro.xacro), the [Gazebo docs](http://gazebosim.org/tutorials?tut=ros_gzplugins#Camera) can help you find out which parameters to set. Don't forget to adapt the [`CameraInfo` publisher](https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/src/gazebo_ros_realsense.cpp#L151-L174).

closed 2018-09-27 by SyrianSpock

q8wwe, 2020-03-06:
how can we add distortion , I need it for depthimage_to_realscan ?