SyrianSpock/realsense_gazebo_plugin #22: add this Camera to iris gazebo has problem.

ifaceyou123, 2018-10-08:
Hi, I am going to add this camera to Iris gazebo, but although it seems that the camera has been added in gazebo, the drone can't take off. This result will be output in the shell.
`[vehicle_spawn_jiaxing_HP_ENVY_15_Notebook_PC_17373_511406443193350206-5] process has finished cleanly
log file: /home/jiaxing/.ros/log/39dbee0e-cae8-11e8-bce5-70188b8817cd/vehicle_spawn_jiaxing_HP_ENVY_15_Notebook_PC_17373_511406443193350206-5*.log
`
No takeoff command line. and there is no connection between quadrotor and QGC.
I was wondering if it was the way I added makes error. I used two methods. The first attempt was to combine the iris and camera directly through the sdf file, and the problem I mentioned above appeared. The second type of addition via xacro files is also problematic. The same error，can not take off, no connection.
Dose somebody successfully add  this camera to IRIS and use them to sitl simulation? I have been trying for a long time to find an answer online, but it has not progressed. I am very sorry that I have raised such a simple question. Thank you for your answer.
Thomas

SyrianSpock, 2018-10-09:
Hi @ifaceyou123, I've never tried to use this plugin in a SITL setup. 
Can you check that your simulation works without the camera and introducing the camera is the source of your problem?

ifaceyou123, 2018-10-10:
Hi @SyrianSpock , Thank you for your reply. Yes, I have tried the Simulation works without the camera and only start the Iris_quadrotor. it works great. but only when I put the camera on it. there will be a problem. now something better is finally there has a connection between quadrotor and QGC. but I get always this problem
`WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro`
and the quadrotor still can not takeoff. 
and I think there is no source problem. because except the EKF problem, there is no other warning or error.

Awesome1995, 2018-11-20:
@ifaceyou123 Hello! have you successfully added this camera to Iris gazebo? I also want to do some simulation using this camera,but failed.I am a beginner in gazebo, I really hope you can help me.
best wishes for you!

ifaceyou123, 2018-11-20:
Hi, I am also failed. Somehow the iris with this camera can takoff and fly
to a waypoints. But there is always a "Manual control lost problem." and I
can't figure out why. So now I am trying to use iris with the  r200 camera,
which is also a default package from firmware px4.
Best wishes for you

SyrianSpock, 2019-03-27:
@Awesome1995 did you manage to solve this problem? I have seen PX4 started using this plugin in their SITL, not sure if it's similar to what you're doing but here is the link https://github.com/PX4/avoidance/pull/245

closed 2019-03-27 by SyrianSpock