SyrianSpock/realsense_gazebo_plugin #23: testCameraInfo Fails

jfeldman24, 2018-10-20:
After following the setup instructions for the kinetic-devel branch and running:
rostest realsense_gazebo_plugin realsense_streams.test

I get the below output:

[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testStream][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testResolution][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testCameraInfo][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testStream][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testResolution][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testCameraInfo][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testStream][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testResolution][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testCameraInfo][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testStream][passed]
[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testResolution][passed]

But then I get the following error:

[realsense_gazebo_plugin.rosunit-realsense_plugin_test/testCameraInfo][FAILURE]-
/home/josh/catkin_ws/src/realsense_gazebo_plugin/test/realsense_streams_test.cpp:205
Value of: encoding_recv
  Actual: "16UC1"
Expected: "mono16"

jfeldman24, 2018-10-20:
The plugin works for the most part, as I can get all 4 image types of camera data in rviz. But when I try to run depth_proc.launch I receive the following error:

[ERROR] [1540068228.270395595]: Failed to load nodelet [/convert_metric] of type [depth_image_proc/convert_metric] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet nodelet_tutorial_math/Plus
[ERROR] [1540068228.270427284]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/convert_metric with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet nodelet_tutorial_math/Plus
[FATAL] [1540068228.270548052, 25.764000000]: Failed to load nodelet '/convert_metric` of type `depth_image_proc/convert_metric` to manager `standalone_nodelet'
[ERROR] [1540068228.284904142]: Failed to load nodelet [/points_xyzrgb] of type [depth_image_proc/point_cloud_xyzrgb] even after refreshing the cache: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet nodelet_tutorial_math/Plus
[ERROR] [1540068228.284944483]: The error before refreshing the cache was: According to the loaded plugin descriptions the class depth_image_proc/point_cloud_xyzrgb with base class type nodelet::Nodelet does not exist. Declared types are  SlamGMappingNodelet nodelet_tutorial_math/Plus
[FATAL] [1540068228.285087367, 25.778000000]: Failed to load nodelet '/points_xyzrgb` of type `depth_image_proc/point_cloud_xyzrgb` to manager `standalone_nodelet'
[convert_metric-2] process has died [pid 13434, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/convert_metric standalone_nodelet image_raw:=/realsense/camera/depth/image_raw image:=/realsense/camera/depth/image_rect __name:=convert_metric __log:=/home/josh/.ros/log/887e8ec4-d4a5-11e8-ac56-c4d987e19808/convert_metric-2.log].
log file: /home/josh/.ros/log/887e8ec4-d4a5-11e8-ac56-c4d987e19808/convert_metric-2*.log
[points_xyzrgb-3] process has died [pid 13435, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb standalone_nodelet --no-bond rgb/camera_info:=/realsense/camera/color/camera_info rgb/image_rect_color:=/realsense/camera/color/image_raw depth_registered/image_rect:=/realsense/camera/depth/image_rect depth_registered/points:=/realsense/camera/depth_registered/points __name:=points_xyzrgb __log:=/home/josh/.ros/log/887e8ec4-d4a5-11e8-ac56-c4d987e19808/points_xyzrgb-3.log].
log file: /home/josh/.ros/log/887e8ec4-d4a5-11e8-ac56-c4d987e19808/points_xyzrgb-3*.log

closed 2018-10-21 by jfeldman24

SyrianSpock, 2018-10-21:
Hi @jfeldman24,
I see you've closed the issue. Did you figure out what was wrong?

SyrianSpock, 2018-10-21:
Good point, I just added `depth_image_proc` as run dependency to the package manifest.