SyrianSpock/realsense_gazebo_plugin #25: Simulating Multiple Cameras

jfeldman24, 2018-10-25:
I am using the kinetic-devel branch with Gazebo 7, but have updated my xacro and launch files to add a prefix to my realsense cameras so multiple can be used in one simulation. I would like to generate multiple instances of the services such that I can map using multiple robots in Gazebo with a RealSense. When the plugin loads I get this error:
` [ERROR] [1540430934.136868574, 0.323000000]: Tried to advertise a service that is already advertised in this node [/realsense/set_camera_info]`

I would like to initialize each sensor in the ns of my individual robots. Is there any support for this/am I doing something wrong?

Thank you in advance.

Maik93, 2018-12-14:
Can you attach a sample launch file that reproduces the error?

GimpelZhang, 2018-12-17:
> Can you attach a sample launch file that reproduces the error?

<launch>
  <arg name="serial_no_camera1"    			default=""/> 			
  <arg name="serial_no_camera2"    			default=""/> 			
  <arg name="camera1"              			default="camera1"/>		
  <arg name="camera2"              			default="camera2"/>		

  <group ns="$(arg camera1)">
    <param
        name="robot_description_1"
        command="$(find xacro)/xacro '$(find realsense_gazebo_plugin)/urdf/rs200_simulation_1.xacro'"
    />
    <node
        name="spawn_model"
        pkg="gazebo_ros"
        type="spawn_model"
        args="-urdf -param robot_description_1 -model rs200_1"
    />
    <node name="state_publisher" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>
  <group ns="$(arg camera2)">
    <param
        name="robot_description_2"
        command="$(find xacro)/xacro '$(find realsense_gazebo_plugin)/urdf/rs200_simulation_2.xacro'"
    />
    <node
        name="spawn_model"
        pkg="gazebo_ros"
        type="spawn_model"
        args="-urdf -param robot_description_2 -model rs200_2"
    />
    <node name="state_publisher2" pkg="robot_state_publisher" type="robot_state_publisher"/>
  </group>
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
    </include>
</launch>

And in rs200_simulation_1.xacro and rs200_simulation_2.xacro, the same urdf file "realsense-RS200.macro.xacro" is implemented. And in "realsense-RS200.macro.xacro", there is a plugin: 
<gazebo>
        <plugin name="realsense_plugin" filename="librealsense_gazebo_plugin.so"/>
        
    </gazebo>

Thanks a lot for your time.

GimpelZhang, 2018-12-17:
> Can you attach a sample launch file that reproduces the error?
I used a launch file to start "rs200_simulation_1.xacro" and "rs200_simulation_2.xacro" (modified from the original file "rs200_simulation.xacro" with model names and camera positions changed)
And in rs200_simulation_1.xacro and rs200_simulation_2.xacro, the same urdf file "realsense-RS200.macro.xacro" is implemented. And in "realsense-RS200.macro.xacro", there is a plugin:
`<gazebo>
        <plugin name="realsense_plugin" filename="librealsense_gazebo_plugin.so"/>
        
    </gazebo>`

Thanks a lot for your time.

GimpelZhang, 2018-12-17:
I used a launch file to start "rs200_simulation_1.xacro" and "rs200_simulation_2.xacro" (modified from the original file "rs200_simulation.xacro" with model names and camera positions changed)
    And in rs200_simulation_1.xacro and rs200_simulation_2.xacro, the same urdf file "realsense-RS200.macro.xacro" is implemented. And in "realsense-RS200.macro.xacro", there is a plugin:
    <gazebo> <plugin name="realsense_plugin" filename="librealsense_gazebo_plugin.so"/>  </gazebo>

Thanks a lot for your time.

GimpelZhang, 2018-12-17:
I used a launch file to start "rs200_simulation_1.xacro" and "rs200_simulation_2.xacro" (modified from the original file "rs200_simulation.xacro" with model names and camera positions changed)
 And in rs200_simulation_1.xacro and rs200_simulation_2.xacro, the same urdf file "realsense-RS200.macro.xacro" is implemented. And in "realsense-RS200.macro.xacro", there is a plugin:
 `   <gazebo> <plugin name="realsense_plugin" filename="librealsense_gazebo_plugin.so"/>  </gazebo>`

Thanks a lot for your time.

SyrianSpock, 2019-03-27:
This should be solved by https://github.com/SyrianSpock/realsense_gazebo_plugin/commit/6230715ef1d5dbabf9d825c90b9604bafb2d9975.

closed 2019-03-27 by SyrianSpock