SyrianSpock/realsense_gazebo_plugin #29: protobuf Version verification failed

TouchDeeper, 2019-03-23:
Hi,
After I run `roslaunch realsense_gazebo_plugin realsense.launch`, error about protobuf version occur:
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.131
[Wrn] [msgs.cc:1655] Conversion of sensor type[depth] not suppported.
[Wrn] [Event.cc:87] Warning: Deleteing a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[libprotobuf FATAL google/protobuf/stubs/common.cc:61] This program requires version 3.4.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/home/wang/ridgeback_cartographer_ws/protobuf/src/google/protobuf/descriptor.pb.cc".)
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  This program requires version 3.4.0 of the Protocol Buffer runtime library, but the installed version is 2.6.1.  Please update your library.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/home/wang/ridgeback_cartographer_ws/protobuf/src/google/protobuf/descriptor.pb.cc".)
Aborted (core dumped)
```
What confuse me is  `Version verification failed in "/home/wang/ridgeback_cartographer_ws/protobuf/src/google/protobuf/descriptor.pb.cc".)`, the workspace mentioned here is not the workspace include the launch file, and if I delete the `/home/wang/ridgeback_cartographer_ws`, the error still be the same. How can I delete the link between the protobuf and `/home/wang/ridgeback_cartographer_ws`?

Maik93, 2019-03-23:
maybe you have sourced that workspace in your `~/.bashrc` file, check if you find something like `source /home/wang/ridgeback_cartographer_ws/devel/setup.bash` and change it to your new workspace location.

SyrianSpock, 2019-03-23:
@TouchDeeper Indeed, you seem to have multiple catkin workspace and your environment is not picking the right one.
Please try @Maik93 's suggestion, and feel free to reopen this issue if you still have problems.

closed 2019-03-23 by SyrianSpock

TouchDeeper, 2019-03-24:
Hi,
I have checked the workspace path by `echo $ROS_PACKAGE_PATH`, the output is below:
`/home/wang/interRealsense/r200_gazebo_plugin_ws/src:/opt/ros/kinetic/share`.
And I have checked the ~/.bashrc, only the `source /opt/ros/kinetic/setup.bash` have not been commented.

TouchDeeper, 2019-03-24:
Can I know which version of protobuf you used?

TouchDeeper, 2019-03-25:
I uninstall the protobuf by `sudo apt-get remove libprotoc-dev`, remember to take a screenshot of the package uninstall with the  protobuf.
and reinstall the protobug by `sudo apt-get install libprotoc-dev`, then reinstall the package uninstall with the protobuf.
The error pass.