SyrianSpock/realsense_gazebo_plugin #34: Is the topic:/r200/camera/depth/image_rect has been aligend to color image?

TouchDeeper, 2019-04-20:
Hello,
Is the topic:/r200/camera/depth/image_rect has been aligend to color image?

SyrianSpock, 2019-04-20:
I don't understand the issue :sweat_smile: 
Is this related to #11 ?

TouchDeeper, 2019-04-20:
The depth image is generated by the infrared camera. The color image is generated by the color camera.
So there exists a transformation between raw depth image and raw color image. 
In general, the depth image will be transformed to the color image.  that is the aligned depth image.
But the topic published by the realsense_gazebo_plugin only has one topic about depth image, so I ask is the topic /r200/camera/depth/image_rect has been aligend to color image.

Maik93, 2019-04-20:
From my personal experience I hazard that depth and RGB images are not
aligned. To rectify one of them on the other you could
use`depth_image_proc` package, but I warn you: it's computationally heavy,
use it if strictly necessary.

TouchDeeper, 2019-04-21:
It seems like that depth and RGB images are not aligned indeed from the image below:
[image]
You can see the color information of the left edge of the coke can have the shifts so it seems like be projected into the floor.

tanyouliang95, 2019-10-09:
Hi, I also faced this alignment issue while trying to use this plugin. The quick fix is to make changes on `realsense-RS200.macro.xacro` urdf file. 
I just made a quick change to both the `depth` and `color` sensor input, making sure both `<pose frame="">` values are the same. This will work, but it's not appropriate. Hopefully there will be future enhancement on this alignment.

[image]

Also, is there any plan to work on realsense d4XX series  plugin?

kmfrick, 2020-03-27:
> I just made a quick change to both the `depth` and `color` sensor input, making sure both `<pose frame="">` values are the same. This will work, but it's not appropriate. 

Why is it not "appropriate"?