SyrianSpock/realsense_gazebo_plugin #37: Two cameras publishing different topics but with the same image

janrosell, 2019-05-15:
I want two cameras in my world.

I've followed the example of multi_rs200_simulation.xacro and modified the location of the cameras there.

In my launch file I load the robot_description parameter and spawn it. Each camera is publishing its topics correctly, e.g.:

/right_r200/camera/color/image_raw
/left_r200/camera/color/image_raw

The problem is that the contents of the image is the same!

Any hint?

Thanks!

ManMan88, 2019-05-26:
There's a bug in RealSensePlugin.cpp - Lines 75 to 87.
The `GetSensor` method takes the name of the sensor as input, but as you can see, it always get the same name. You need to modify it to use the name of the sensors - in your case, you could use `left_****` and 'right_****'.

The easiest way I found to do it is:
In the realsense macro, in the plugin tag, insert a prefix tag (or name it differently, it doesn't matter), like so:

```
 <!-- Load plugin to the model ("robot" in urdf format)-->
  <gazebo>
      <plugin name="${prefix}r200" filename="librealsense_gazebo_plugin.so">
          <prefix>{$prefix}</prefix>
      </plugin>
   </gazebo>
```

And also, for each sensor name, add the prefix. For example:
`<sensor name="color" type="camera">`
change to:
`<sensor name="{$prefix}color" type="camera">`

Then modify the RealSensePlugin.cpp:
```
  // set camera prefix
  std::string prefix = "";
  if (_sdf->HasElement("prefix"))
    prefix = _sdf->Get<std::string>("prefix");

  // Sensors Manager
  sensors::SensorManager *smanager = sensors::SensorManager::Instance();

  // Get Cameras Renderers
  this->depthCam = std::dynamic_pointer_cast<sensors::DepthCameraSensor>(
                                smanager->GetSensor(prefix + DEPTH_CAMERA_NAME))
                                ->DepthCamera();
  this->ired1Cam = std::dynamic_pointer_cast<sensors::CameraSensor>(
                                smanager->GetSensor(prefix + IRED1_CAMERA_NAME))
                                ->Camera();
  this->ired2Cam = std::dynamic_pointer_cast<sensors::CameraSensor>(
                                smanager->GetSensor(prefix + IRED2_CAMERA_NAME))
                                ->Camera();
  this->colorCam = std::dynamic_pointer_cast<sensors::CameraSensor>(
                                smanager->GetSensor(prefix + COLOR_CAMERA_NAME))
                                ->Camera();
```

SyrianSpock, 2019-12-04:
@ManMan88 Are you willing to make this fix into a PR? I can review and merge to solve this issue