SyrianSpock/realsense_gazebo_plugin #41: Test Fails

GinaJemy, 2019-10-03:
I'm using ROS kinetic with gazebo 7
running the unit tests fails with the followinr error
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

Also when launching the realsense.launch file 
this warning appears : Conversion of sensor type[depth] not suppported.
and gazebo doesn't open

SyrianSpock, 2019-12-04:
are you using the branch `kinetic-devel` with ROS kinetic?

q8wwe, 2020-03-03:
no solution for this problem ???

SyrianSpock, 2020-03-03:
Trying to understand it locally with little success, sorry. Also [busy IRL with my treatment](https://www.syrianspock.com/2019/10/22/enter-the-glioblastoma/) so not fully working on the issue unfortunately

SyrianSpock, 2020-03-03:
I need to fix this and set up proper CI for this project

SnowCarter, 2020-03-03:
Can I catkin build in ubuntu 18.04?

SyrianSpock, 2020-03-04:
Yes if you install https://catkin-tools.readthedocs.io/en/latest/

q8wwe, 2020-03-04:
I wish you a good health and full recovery very soon .
I'm trying for a whiLe now to make realsense work with gazebo but no luck until now .
I'm working on a wheeled robot and will appreciate any hint ,shall I drop the idea on working with simulation and shift to real robot ? Or realsense can use with gazebo and simulated .
You are an important member in the society and your contribution have a huge value .I wish if you can supporting is on this issue .

Best regards

SnowCarter, 2020-03-04:
I have built it in ubuntu 18.04 successfully. Great.

SyrianSpock, 2020-03-04:
> shall I drop the idea on working with simulation and shift to real robot ?

I think simulation is a good tool

> Or realsense can use with gazebo and simulated .

This plugin can be used to simulate an R200 camera

Pao-Te, 2021-02-24:
> I have built it in ubuntu 18.04 successfully. Great. If there are the realsense T265's plugin, it will be much better. Thank you very much.

Hi, I am a beginner on ROS, and I work on ubuntu 18.04.
I follow the instructions to build the code, but I fail.
Could you help me??

 In terminal,

$ cd catkin_ws/src/
$ git clone https://github.com/SyrianSpock/realsense_gazebo_plugin
$ catkin build realsense_gazebo_plugin
$ rostest realsense_gazebo_plugin realsense_streams.test

Test fails,

... logging to /home/rsl/.ros/log/rostest-rsl-ROG-Zephyrus-G14-GA401IV-GA401IV-3968.log
[ROSUNIT] Outputting test results to /home/rsl/.ros/test_results/realsense_gazebo_plugin/rostest-test_realsense_streams.xml
[ INFO] [1614166781.837500041]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1614166781.840298984]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1614166782.193158810]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1614166782.210460164]: Physics dynamic reconfigure ready.
[ INFO] [1614166782.490344012]: Realsense Gazebo ROS plugin loading.

RealSensePlugin: The realsense_camera plugin is attach to model realsense_camera
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/launch.py", line 689, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/rsl/.ros/log/rostest-rsl-ROG-Zephyrus-G14-GA401IV-GA401IV-3968.log

Moreover, 

roslaunch realsense_gazebo_plugin realsense.launch # in terminal 1
roslaunch realsense_gazebo_plugin depth_proc.launch # in terminal 2

I can not see depth points in rviz.

Could you tell me your solutions?

Thanks a lot!!!