SyrianSpock/realsense_gazebo_plugin #42: No points published for kinetic devel

CesMak, 2019-11-13:
Hey there,

just tested the kinetic-devel branch unittests:

```
ostest realsense_gazebo_plugin realsense_streams.test
... logging to /home/mlamprecht/.ros/log/rostest-mlamprecht-pc-8945.log
[ROSUNIT] Outputting test results to /home/mlamprecht/.ros/test_results/realsense_gazebo_plugin/rostest-test_realsense_streams.xml
[ INFO] [1573634895.473643669]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1573634895.474245803]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1573634895.957811694]: Realsense Gazebo ROS plugin loading.

RealSensePlugin: The realsense_camera plugin is attach to model realsense_camera
[ INFO] [1573634896.075718407, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1573634896.096323756, 0.042000000]: Physics dynamic reconfigure ready.
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

```

If I try to see the point cloud, no points are published. This is what I did:

```
roslaunch realsense_gazebo_plugin realsense.launch # in terminal 1
roslaunch realsense_gazebo_plugin depth_proc.launch # in terminal 2
Then open Rviz, and display the /realsense/camera/depth_registered/points topic, you should see something like this 
```

anubhav1772, 2019-11-15:
Hey, have you able to solve above error? I am also getting the same error.

SyrianSpock, 2019-12-04:
are you using the branch `kinetic-devel`?

SyrianSpock, 2019-12-04:
Are you using ROS kinetic with Gazebo 7?

MoTahoun, 2020-01-31:
@SyrianSpock I got the same error and yes I am using ROS kinetic with Gazebo 7.
Is there any solution?

Thanks in advance

SyrianSpock, 2020-03-04:
This is weird do you still get the issue on the current master branch?

zzy5510, 2020-03-21:
Hi, guys, i am using ubuntu16.04+ROSkinetic+gazebo7.0, and it works normally.I used to have the same problem and now i solved it. I found it is because i forgot to run depth_proc.launch. Make sure you have run it and the point topic is /r200/camera/depth_registered/points.

danem, 2020-04-03:
As for failing the unit tests:

I'm running melodic and using the master branch and had the same failure as OP. In [realsense_streams_test.cpp](https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/test/realsense_streams_test.cpp#L10) I changed the topic names to start with `realsense_plugin` and now it passes every test except the [last one](https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/test/realsense_streams_test.cpp#L202).

I'm still unable to get the pointcloud working however. I've tried using `rosrun tf static_transform_publisher 0 0 0 0 0 0 origin depth 1` to create the tf frames necessary for Rviz, without any luck.