SyrianSpock/realsense_gazebo_plugin #43: Conversion of sensor type[depth] not supported.

anubhav1772, 2019-11-15:
Hi, 
I am a newbie in Ros and trying to run `realsense_gazebo_plugin` but getting some warnings. On running following command `roslaunch realsense_gazebo_plugin realsense.launch`, I am getting below warning messages:
```
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call

RealSensePlugin: The rs_camera plugin is attach to model realsense_camera
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
[ INFO] [1573849778.403345017, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1573849778.432398530, 0.041000000]: Physics dynamic reconfigure ready.
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
[Wrn] [msgs.cc:1852] Conversion of sensor type[depth] not supported.
```
I don't have any idea why I am getting above warnings. But, when I ignored these warnings and run following command: `roslaunch realsense_gazebo_plugin depth_proc.launch` and then start rviz and added pointcloud2 display and set topic `/r200/camera/depth_registered/points`, I am not getting output. Could someone please guide me on this? I would really appreciate your help.

grafoteka, 2019-11-27:
Hello @anubhav1772, could you solve it? 
I'm having the same problem as you.

Thank you

anubhav1772, 2019-11-27:
Hey @grafoteka ,
Unfortunately, I wasn't able to solve it yet.

SyrianSpock, 2019-11-27:
Weird. What ROS and Gazebo versions do you use?

grafoteka, 2019-12-03:
Hi @SyrianSpock, I'm using ROS Kinetic and Gazebo 7.15

SyrianSpock, 2019-12-03:
Dumb question: @grafoteka are you using the branch `kinetic-devel`?

grafoteka, 2019-12-03:
Hi @SyrianSpock , dumb questions are always the best ones!

I forgot to specify the kinetic branch when clone the repo. Now is working perfectly!! 

Thank you very much for the repo!!

SyrianSpock, 2019-12-04:
Please try the same @anubhav1772 and please re-open if it persists

closed 2019-12-04 by SyrianSpock

SnowCarter, 2020-06-26:
I'm sorry, could you please share what you have done? Where should we specify the kinetic branch?

HartmannSa, 2020-10-20:
Hi, you should specifiy the branch while cloning the repo, e.g with the command:
git clone -b kinetic-devel https://github.com/pal-robotics/realsense_gazebo_plugin.git

I use Gazebo 9.0 and ros-melodic on Ubuntu 18.04.5 LTS and I get the same error 
[msgs.cc:1852] Conversion of sensor type[depth] not supported.

Due to other reasons the solution "downgrading to Gazebo 7 and use ros-kinetic" is not working for me.
Morover I cloned the correct version (in my case melodic-devel).
So is there a solution for this error for Gazebo 9 and ros-melodic?