SyrianSpock/realsense_gazebo_plugin #44: Kinetic + Gazebo9

QING-ML, 2019-11-20:
could this Realsense_plugin rightly run in kinetic + Gazebo 9? now when I run this plugin in Gazebo 9 . There is no rostopic about /realsense

SyrianSpock, 2019-11-20:
It should I will verify this and report back

SyrianSpock, 2019-12-04:
@QING-ML are you using the branch `kinetic-devel`?

QING-ML, 2019-12-04:
Yes i use kinetic devel

SyrianSpock, 2019-12-04:
Ok I will attempt to replicate the bug locally

SyrianSpock, 2019-12-04:
Note: I only test on ROS Kinetic with Gazebo 7 the officially supported version AFAICT http://gazebosim.org/tutorials/?tut=ros_wrapper_versions

SyrianSpock, 2019-12-04:
`kinetic-devel` branch should probably be called `gazebo7` instead, so maybe try the `master` branch since you're on Gazebo 9 please ? (https://github.com/SyrianSpock/realsense_gazebo_plugin/commit/57d824fb3ba870252ff9e1841430377dc55a5b4e)

SyrianSpock, 2020-03-04:
Since https://github.com/SyrianSpock/realsense_gazebo_plugin/commit/5e332862827674874af43844fcfccfe11a76af95 it should work

closed 2020-03-08 by SyrianSpock