SyrianSpock/realsense_gazebo_plugin #46: Test Failed ROS-kinetic Gazebo-7

MoTahoun, 2020-01-31:
I'm using ROS kinetic with gazebo 7 and cloned the folder from "`kinetic-devel`" branch.
This is the command that I run,
`rostest realsense_gazebo_plugin realsense_streams.test`
and here are the output,
```
... logging to /home/tahoun/.ros/log/rostest-mTahoun-Precision-5520-24014.log
[ROSUNIT] Outputting test results to /home/tahoun/.ros/test_results/realsense_gazebo_plugin/rostest-test_realsense_streams.xml
[ INFO] [1580468558.084234952]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1580468558.084696070]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...

RealSensePlugin: The rs_camera plugin is attach to model realsense_camera
[ INFO] [1580468558.856333616, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1580468558.907369463, 0.071000000]: Physics dynamic reconfigure ready.
[Testcase: testrealsense_plugin_test] ... ERROR!
ERROR: max time [60.0s] allotted for test [realsense_plugin_test] of type [realsense_gazebo_plugin/test_realsense_streams]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/tahoun/.ros/log/rostest-mTahoun-Precision-5520-24014.log
```

Also, here is attached the log file,
[rostest-mTahoun-Precision-5520-24014.log](https://github.com/SyrianSpock/realsense_gazebo_plugin/files/4139006/rostest-mTahoun-Precision-5520-24014.log)

Thanks in advance and waiting for your reply.
Regards,

SyrianSpock, 2020-02-03:
Are you using the kinetic-devel branch?

MoTahoun, 2020-02-04:
@SyrianSpock Yes, I am using the `kinetic-devel` branch

SyrianSpock, 2020-03-04:
duplicates to #41

closed 2020-03-04 by SyrianSpock