SyrianSpock/realsense_gazebo_plugin #49: Why are there 2 IR cameras and 4 Gazebo sensors?

felixvd, 2020-03-09:
From what I understand, the SR300 has one IR projector, one IR camera and one RGB camera. However, in the [URDF](https://github.com/SyrianSpock/realsense_gazebo_plugin/blob/master/urdf/realsense-RS200.macro.xacro) I see these 4 joints/links, and a Gazebo sensor that is spawned for each of them:

```
    <link name="${prefix}color" />
    <link name="${prefix}depth" />
    <link name="${prefix}ired1" />
    <link name="${prefix}ired2" />
```

What do the `ired1` and `ired2` cameras mean? Shouldn't there only be one? Thanks in advance.

PS: I am asking because I would like to make a URDF for the SR305, but I am confused since there is only one IR sensor in the CAD file.

SyrianSpock, 2020-03-09:
We simulate the R200 which has a stereo IR rig, so that's two IR cameras.

closed 2020-03-09 by SyrianSpock

felixvd, 2020-03-10:
Thanks. I like when dumb questions have simple answers :)