SyrianSpock/realsense_gazebo_plugin #50: Is there any kinetic branch package?

zzy5510, 2020-03-21:
the master branch seems not works rightly on my environment.
the move_group node  always exits

SyrianSpock, 2020-03-21:
Master should work can you post the error please?

zzy5510, 2020-03-21:
emmm...
the plugin works rightly. And i can get cloudpoint if I run as you said in Readme.md.
However, when i uses this plugin in moveit, which means moveit will listen to this topic and transform it to Octomap, so moveit can feel the obstacle in the environment, and it fails to roslaunch ur5_planning_executin_example_execution.launch.
the error is:
[ERROR] [1584798236.109192037, 1062.893000000]: Transform error: Lookup would require extrapolation into the past.  Requested time 1038.528000000 but the earliest data is at time 1061.233000000, when looking up transform from frame [gripper_finger1_knuckle_link] to frame [color]
[ERROR] [1584798236.109424738, 1062.893000000]: Transform cache was not updated. Self-filtering may fail.

And the same config with a kinect camera works normally.
I think this may be too complicated. Can you get what i said?
Thankyou anyway.

SyrianSpock, 2020-03-21:
Are you using sim time?

zzy5510, 2020-03-22:
i have  tried both ur5_planning_executin_example_execution.launch and depth_proc.launch use sim time or don't use sim_time. in the former case, the error before will show  once, while in the latter case, the error will show preiodicly.
That is supper strange.
And i find if i run  ur5_planning_executin_example_execution.launch before depth_proc.launch, it has no error but i get no pointcloud.even the topic of  pointcloud exist and movegroup has subscripted it.

SyrianSpock, 2020-03-22:
Please share your launch files

zzy5510, 2020-03-22:
this is launch_simulation.launch

[launch_simulation.txt](https://github.com/SyrianSpock/realsense_gazebo_plugin/files/4364927/launch_simulation.txt)

[main.txt](https://github.com/SyrianSpock/realsense_gazebo_plugin/files/4364929/main.txt)
[urwh_planning_example_execution.txt](https://github.com/SyrianSpock/realsense_gazebo_plugin/files/4364930/urwh_planning_example_execution.txt)
 These are three launch files.
I will first launch main.launch to run gazebo, and then run urwh_planning_example_execution.launch to run rviz.

zzy5510, 2020-03-28:
Hi guys and SyrianSpock, thanks for your help and it turns out to be a time problem. Now i use sim time in all my nodes and this error disappear. Unlucky, i got another error:

[ERROR] [1585381836.248724327, 1517.576000000]: Transform error: "shoulder_link" passed to lookupTransform argument source_frame does not exist. 
 
And I can not find this mysterious source_frame in every urdf. What does this mean?