SyrianSpock/realsense_gazebo_plugin #51: depthimage_to_laser ' Cannot call rectifyPoint when distortion is unknown.'

q8wwe, 2020-03-21:
plugin worked for me but I cant use it with depthimage to laser :

Could not convert depth image to laserscan: Cannot call rectifyPoint when distortion is unknown.

please advice how to solve the problem

SyrianSpock, 2020-03-21:
Which ROS/Gazebo do you use?

q8wwe, 2020-03-21:
I used Ros Kinetic with Gazebo 7

drjsmith, 2020-10-10:
@q8wwe @SyrianSpock I encountered the same problem some time ago. The source of the error is that rectifyPoint requires a distortion model and no distortion model is being specified in the CameraInfo message. The `fix-distortion-model` branch of this [repo](https://github.com/ivaROS/realsense_gazebo_plugin) contains my fix. This [diff](https://github.com/ivaROS/realsense_gazebo_plugin/commit/22aeb9fd9296542d84ad946e97381e4370186df4#diff-f44a7df296ed73fcea37f511b9527049) shows how the distortion model (which in this case indicates no distortion) can be set.